Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #2: [question] Exploits compatibility ESP32

BigMaker42, 2021-11-02:
Hello guys !
Hope you're doing well ? 

I would like to know if I could run this exploits with my ESP32 NodeMCU ? 
Sorry but i'm quite a noob aha

Thank you in advance for your help !
Best regards 🙂

mobier, 2021-11-03:
I tried and found that it couldn't work. Maybe the serial port chip on NodeMCU is CP2102, and the maximum baud rate of the chip is 3M

Matheus-Garbelini, 2021-11-03:
Hi @mobier , unfortunatelly, FT2232H is required. The PoC operates at 4M baud and requires latency_timer set to 0 to ensure low latency communication with the board (~125us).

Currently, only ESP32 boards ESP-WROVER-KIT or ESP-ETHERNET-KIT have such FT2232H integrated.
The alternative is to manually connect a ESP-PROG board to your ESP32. ESP-PROG also contains a FT2232H

colarocker, 2021-11-11:
Hello @Matheus-Garbelini , i'd like to ask if you could provide additional notes using the esp-prog, i tried it using the braktooth-scanner, but it won't connect to a port. 

The host system i'm using is ubuntu 18.04 running in a vmware workstation.

I tried using a ESP32-devkit v1, flashed the firmware using the usb port of the esp (and tried a jtag-flash instead a serial-flash with the esp-prog, both using the vm), the flashing suceeds with both attempts but the scan won't. 
I connected the following ports TDI: GPIO12, TCKL: GPIO13, TMS: GIO14, TDO: GPIO15. 
Both the esp32 and the esp-prog are connected via usb (2x usb3.0 port, vmwares usb compatibility is set to 3.1).

`[ESP32BT] Probing /dev/ttyUSB0 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Probing /dev/ttyUSB1 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Probing /dev/ttyUSB2 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Firmware version not detected
Serial port could not open.`

could there be something i'm missing? thanks for the help!

Matheus-Garbelini, 2021-11-12:
Hi @colarocker have you also connected the txd/rxd pins from ESP-prog 2x3 connector to your ESP32-devkit v1? 
The jtag can be used to program the kit, but only the serial pins are used to communicate with ESP32. Note that ESP32-devkit v1 already has a USB connector, but the idea is that you bypass that by using ESP-PROG TX/RX pins exposed on the 2x3 connector.
So in short, do not connect the ESP32-devkit v1 onboard USB port to your PC, because that would conflict with your ESP-PROG 2x3 connection.

Ideally you should only have the following USB ports listed under your /dev/ directory:
`/dev/ttyUSB0` -> This corresponds to the JTAG via SPI. As pointed by you, this works fine. Once you flash the firmware, this port disappears as platformio calls openocd which switches FT2232H channel A mode from serial to SPI.
`/dev/ttyUSB1` -> This is the serial port exposed on ESP-PROG 2x3 connector. That's what the exploit tries to look for.

**Note:**
Follow this recommendation: https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks/issues/3#issuecomment-963734219

It disables auto search so you can specify `/dev/ttyUSB1` when launching the PoC

Anything else, let me know.
Regards.

mringwal, 2021-11-16:
Hi @Matheus-Garbelini ! 

I (finally) wanted to give it a try but I get stuck with the communication setup after flashing the ESP32 with the provided tool as described in the README.md.
- With an ESP32 Thing, I got 'cannot open' port (probably because of the 4 mb/s >  FT231XQ)
- With an u-blox evk-nina-w10t, ESP32 + FTDI4232HQ (quad high-speed), I get 'no response' / 'Firmware version not detected' on all 4 ports.

How can I debug/fix this? Would it help to buy the ESP-WROVER-KIT?

I wanted to try the AU Rand exploit with the CC2564. From the description, it seems that it could also work with slower UART as there's no need to intercept LMP PDUs.

Thanks for the easy setup tools!

Matheus-Garbelini, 2021-11-17:
Hi @mringwal does your ESP32 Thing or u-blox evk-nina-w10t
works with the sniffer firmware instead? https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer
Indeed there is the requirement of 4Mbps which FT231XQ does not support, however I'm not sure why 
FTDI4232HQ didn't work for your case given that its datasheet reports 12M max baud.
We have only validated the PoC with [ESP-WROVER-KIT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html), or [ESP32-Ethernet-Kit](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html), or ESP-PROG + ESP32.

Regarding the AU Rand. Yes, this attack can be triggered by just injection rather than interception.
In this case, you can try using the sniffer firmware with the PoC, but you'd need to change the parameter `SerialBaudRate`
from 4000000 to 921600 in `configs/bt_config.json`

There is a way to debug the serial port to see whether the firmware ever receives anything.
If you send 0xEE (in binary format) at 4M/921600 baud, the firmware replies with a version string "v1.4.1".
Normally, if you can program the firmware via any serial port of a supported FTDI chipset, then it should work.
Except in the case that FTDI removes support for latency_timer=0 in this quad chip that you have.

If you cannot make the current firmware or the sniffer firmware with this PoC, let me know and I'll add a different PoC for AU rand flooding on the sniffer repository using python code.

Matheus-Garbelini, 2021-11-23:
Hi @mringwal ,
were you able to replicate AU_RAND flooding vulnerability on CC2564?

Regards

mringwal, 2021-11-23:
@Matheus-Garbelini  Thanks for asking. Sorry to not report earlier. Yes, I got the AU_RAND flooding to work on the borrowed WROVER board. It worked right away and it's quite reliable - thanks for providing the easy to use PoC.

With this, I was able to improve/fix BTstack's logic to  do a power cycle. When using a watchdog in the app (e.g. sending Read BD ADDR), the app can repeatedly power cycle as long as the AU_RAN flooding attack is active and be ready when it stops. So that's done, great!

I've  tried the sniffer firmware but wasn't able to get it working on ESP32 Thingy or  u-blox evk-nina-w10t. I didn't had an idea how to send a single 0xEE on mac or linux quickly (without writing actual code to do so).

Matheus-Garbelini, 2021-11-23:
@mringwal glad to know that the PoC helped to improve bluekitchen!!!

I'll check the sniffer issue with ESP32 Thingy when I get the chance.
In any case, here's a sample python3 script in case you try this on your mac:

<details>
  <summary>ESP32 Sniffer Firmware Version Tester</summary>
  
 
```python3
#!/usr/bin/env python3
import serial
import termios
import fcntl
from threading import Thread
from time import sleep
import struct

ser = serial.Serial("/dev/ttyUSB1", 4000000, rtscts=0,
                    dsrdtr=0, timeout=20)

def set_dtr_rts(ser, dtr, rts):
    # Linux only for now
    import termios
    import fcntl
    fd = ser.fileno()

    TIOCMGET = getattr(termios, 'TIOCMGET', 0x5415)
    TIOCMSET = getattr(termios, 'TIOCMSET', 0x5418)
    TIOCM_DTR = getattr(termios, 'TIOCM_DTR', 0x002)
    TIOCM_RTS = getattr(termios, 'TIOCM_RTS', 0x004)
    TIOCM_zero_str = struct.pack('I', 0)

    TIOCM = fcntl.ioctl(fd, TIOCMGET, TIOCM_zero_str)[0]
    TIOCM &= ~(TIOCM_DTR | TIOCM_RTS)
    if dtr:
        TIOCM |= TIOCM_DTR
    if rts:
        TIOCM |= TIOCM_RTS

    TIOCM_str = struct.pack('I', TIOCM)
    fcntl.ioctl(fd, TIOCMSET, TIOCM_str)

def send_msg():
    while True:
        ser.write(b'\xEE')
        print('0xEE sent')
        sleep(1)

Thread(target=send_msg, daemon=True).start()

try:
    while True:
        m = ser.readline()
        print(m)
        if m and len(str(m).split('.')) == 3:
            print('SUCCESS: Version detected, firmware OK!')
        else:
            print('ERROR: Something went wrong, no firmware version detected!')
except:
    print('\n')

```
  
</details>

colarocker, 2021-11-26:
Thanks so much for the info, i finally got it working.
Confusing naming here was the pin-naming in the official documentation for the esp-prog, worth to mention to connect ESP_RX (esp-prog) to RX0 (esp32) and ESP_TX (esp-prog) to TX0 (esp32). 

Finally, [i followed these instructions to set access permissions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html?highlight=jtag#linux) for the USB-ports, i think this step was crucial.

best wishes!

BigMaker42, 2022-02-08:
Hello guys, i'm back ! 

Thank you for your answers @Matheus-Garbelini ! I just bought an ESP32-prog for my ESP32 nodeMCU, once I receive and test it, I would do a step by step guide to install the PoC with these components (ESP32-Prog + ESP32 nodeMCU). 

Here is the datasheet for the ESP32 nodeMCU that i own : [Datasheet ESP32 NodeMCU WLANWiFi Development Board](https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_AZ-Delivery_Vertriebs_GmbH_10f68f6c-a9bb-49c6-a825-07979441739f.pdf?v=1598356497)
Here is the datasheet for the ESP32-Prog : [Datasheet ESP32-PROG](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html)

I think it might be a better option, especially compared to the price of the ESP-WROVER-KIT or the ESP32-Ethernet-Kit.😊

Matheus-Garbelini, 2022-06-10:
Hi @BigMaker42 sorry for the big delay in getting back to this.
Thanks a lot for your wonderful tutorial.
Can I include your tutorial in the main readme page or you prefer to open a PR.

Regards.

BigMaker42, 2023-07-28:
Hi @Matheus-Garbelini,
Sorry for answering so lately ! 
You must check if it's still up to date but yes can post it on the main readme page ! 😄 

Have a great day !

closed 2023-07-28 by BigMaker42