Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #3: BUG: The host port can not be specified?

JianwenLi-GS, 2021-11-09:
the bin/bt_exploiter still try to probe ttyUSB0/1/2, even the ttyUSB3 has been specified by '--host-port' arg.

```bash
 sudo bin/bt_exploiter --host-port=/dev/ttyUSB3 --target=xx:xx:xx:xx:xx:xx --exploit=au_rand_flooding 
Logical Cores: 4
No SMT support
Assigned CPUSET:
CPU 0 Allowed
CPU 1 Allowed
CPU 2 Allowed
CPU 3 Allowed
sched_setscheduler: Current process set to realtime (RR Scheduler)
Thread priority is 99
/proc/sys/kernel/sched_rt_runtime_us = -1
Enabling Core dump: ulimit -c unlimited
Loading Model...
Model Loaded!
[Modules] Loading C++ Modules...
GCC not found, modules won't be compiled from source.
[Modules] --> duplicated_iocap.so loaded
[Modules] --> wrong_encapsulated_payload.so loaded
[Modules] --> invalid_timing_accuracy.so loaded
[Modules] --> invalid_max_slot.so loaded
[Modules] --> duplicated_encapsulated_payload.so loaded
[Modules] --> sdp_unkown_element_type.so loaded
[Modules] --> sdp_oversized_element_size.so loaded
[Modules] --> invalid_feature_page_execution.so loaded
[Modules] --> repeated_host_connection.so loaded
[Modules] --> paging_scan_disable.so loaded
[Modules] --> feature_req_ping_pong.so loaded
[Modules] --> feature_response_flooding.so loaded
[Modules] --> au_rand_flooding.so loaded
[Modules] --> lmp_overflow_2dh1.so loaded
[Modules] --> lmp_auto_rate_overflow.so loaded
[Modules] --> noncompliance_invalid_stop_encryption.so loaded
[Modules] --> noncomplicance_duplicated_encryption_request.so loaded
[Modules] --> knob.so loaded
[Modules] --> lmp_overflow_dm1.so loaded
[Modules] --> invalid_setup_complete.so loaded
[Modules] --> truncated_lmp_accepted.so loaded
[Modules] --> truncated_sco_link_request.so loaded
[Modules] --> lmp_max_slot_overflow.so loaded
[Modules] 23/23 Modules Compiled / Loaded
[Modules] au_rand_flooding configured and ready!
[ESP32BT] HCI Bridge ON: /dev/pts/4
[ESP32BT] Discovering Serial Port
[ESP32BT] Probing /dev/ttyUSB0 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Probing /dev/ttyUSB1 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Probing /dev/ttyUSB2 at 4000000 baudrate...
[ESP32BT] No response
[ESP32BT] Probing /dev/ttyUSB3 at 4000000 baudrate...
[ESP32BT] Got valid response from /dev/ttyUSB3
[ESP32BT] Firmware version: 1.3.0
```

Matheus-Garbelini, 2021-11-09:
Hi @JianwenLi-GS

You can disable this behavior by changing `configs/bt_config.json` and set `"SerialAutoDiscovery"` property to **false**

JianwenLi-GS, 2021-11-09:
> Hi @JianwenLi-GS
> 
> You can disable this behavior by changing `configs/bt_config.json` and set `"SerialAutoDiscovery"` property to **false**

It works, Thanks

Larryluk2022, 2022-01-18:
Hi Matheus, 

i got same issue and disable this behavior as you suggested, to change the property of "serialAutoDiscovery" to false. 

But for my case, it was not working, and it shown the "module section" were okay, 

failed on the ESP32BT, which can't be detected. 
I have no idea where is going wrong, as it was okay at initial scanning and launching.
At least, i got one successful trial before 

[Modules] 23/23 Modules Compiled / Loaded
[Modules] au_rand_flooding configured and ready!
[ESP32BT] HCI Bridge ON: /dev/pts/1
[ESP32BT] Firmware version not detected
Serial port could not open.
[Monitor] Disabled
[ESP32BT] Firmware version not detected
[ESP32BT] Firmware version not detected
[ESP32BT] Firmware version not detected
[ESP32BT] Firmware version not detected