Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #6: Missing source code?

jiska2342, 2021-12-20:
Hi,

thanks for publishing the tooling for the ESP32 :)

I finally had some time to take a look into the source code and realized that only the source for the exploits is available in `wdexploiter/modules/exploits`. However, the source code for `wdexploiter/bin/bt_exploiter` is missing. Thus, the ESP32 tooling allows to configure BLE packets according to the definitions but not adding any other undefined behavior.

Did I just miss the source code? It would be really interesting to have a binary patcher for the ESP32 firmware.

Matheus-Garbelini, 2021-12-22:
Hi @jiska2342, thanks for reaching out.
The `bt_exploiter` is similar to the [esp32 active sniffer](https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer) as it connects to the board and receives LMP packets.
However, it additionally loads the user-made code from `exploits` folder. So the binary patcher is not there.

1. To add new new BR/EDR undefined cases, the user can create an arbitrary `new_case_name.cpp` file in `wdexploiter/modules/exploits/` following API's template from other exploits and  `bt_exploiter` will compile, load and run the new case (exploit) when it starts the next time with `--exploit=new_case_name.cpp` switch.
2. If you mean adding undefined behavior direct to the firmware.
The firmware is yet to be open-sourced on the [esp32 active sniffer repo.](https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer). It has been postponed to early February (sorry about that), but I can send you the source by email if you don't mind the lack of documentation. The patching process is integrated into the normal user-code build tool (platformio) and the generated ELF file.
4. A pseudo terminal is created on /dev/pts/x. This is to emulate a HCI serial port. You can then connect to any BT stack you want (e.g., bluez or other blukitchen profile programs) to guide the LMP communication.
3. `bt_exploiter` cannot be open-sourced for now due to Keysight collaboration requirements, but the ESP32 firmware source code and binary patching framework can.

Regards.

AKoo7, 2022-10-10:
Hi Matheus,

Thanks for sharing POC code, but i think firmware source for esp32 active sniffer is still not available. Is there any plan to make that code publically available?

Thanks and regards,

Matheus-Garbelini, 2022-10-10:
Hi @AKoo7 the firmware source code is pubicly available here:
https://github.com/Matheus-Garbelini/esp32_firmware_patching_framework

AKoo7, 2022-10-10:
Thanks Matheus, sorry i missed that somehow.