Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #7: Scanning for blutooth targets finds fewer than expected

charliecryer, 2022-05-18:
Hello Braktooth team! 

My goal is to use the PoC's feature response flooding to crash a DOSS SoundBox Wireless Portable Bluetooth Speaker I've got. I'm following through the README included in the proof of concept, and I'm running into an issue. I can't seem to find the speaker when I scan with `sudo bin/bt_exploiter --scan`.

Here is an excerpt from the output when calling `sudo bin/bt_exploiter --scan`:
```
[ESP32BT] Probing /dev/ttyUSB1 at 4000000 baudrate...
[ESP32BT] Got valid response from /dev/ttyUSB1
[ESP32BT] Firmware version: 1.3.0
[ESP32BT] LMP Sniffing ENABLED
[ESP32BT] TX Packet interception ENABLED
[ESP32BT] RX Bypass ENABLED
[ESP32BT] [!] Bypass on Demand DISABLED
[ESP32BT] [!] Role Switch ENABLED
[ESP32BT] Own BDADDR set to 50:4a:35:a0:6c:6f
[ESP32BT] Measuring UART Latency...
[ESP32BT] USB Latency:142 us [OK]
Serial port /dev/ttyUSB1@4000000 opened
BT Scanning Started (Inquiry)...
[ESP32BT] BDAddress: f8:59:71:63:b9:d2, Name: ACOMBSR-5520, RSSI: -75, Class: Laptop
[ESP32BT] BDAddress: 90:61:ae:87:e0:32, Name: charlie-HP-ZBook-Studio-G4, RSSI: -57, Class: Laptop
[ESP32BT] BT Scanning Finished, got 2 result(s).
Fuzzer Closed
```
The fundamental problem is that the scanning function is finding far fewer BT devices than there are actually available. My phone will see 6+ potential connections while scanning shows only 2. 

So far, I have tried 2 fixes, neither of which worked: Putting the speaker into pairing mode, and sourcing the speaker's BDAddress from somewhere else. Putting the speaker in pairing mode did not make a difference. I tried to use the BDAddress my computer gives me when I connect to the speaker, but when attempting the exploit, it seems the ESP32 cannot connect the speaker, printing:
```
BTstack up and running at 3B:41:CF:56:EF:E5
[Baseband] TX --> FHS
SDP query failed 0x04, retrying...
[Baseband] TX --> FHS
[Timeout] No Response received for 30 seconds
[Timeout] Target is not responding, check if target is still alive...
```

I'm a newbie to security and bluetooth stuff, but the report and docs you guys have made for braktooth have been immensely helpful. Many thanks for all the work you have done and for any help with this issue.

Sincerely,
Charlie

Matheus-Garbelini, 2022-05-19:
Hi @charliecryer 
The scanning function seems to not be that complete, therefore I recommend you to find the BDAddress of your target by other means instead fo using --scan. You can get the BDAddresses via some smartphone app or getting it via your laptop, etc.

charliecryer, 2022-05-19:
Hello @Matheus-Garbelini,

Thanks for the advice! I was able to successfully connect to my speaker. I used my laptop to find the address, though running the exploit gave me trouble. It turns out that the speaker does not accept multiple bluetooth connections, so my esp32 couldn't establish a connection while my phone was playing music on it.

I would like to replicate the feature response flooding exploit demonstrated [here](https://youtu.be/AekAMurR5Kk), so I've been looking around for vulnerable speakers that accept multiple connections. Do you have any advice for identifying the BT SoCs for consumer speakers? The user manuals and product specs on the website usually don't mention the BT chip they have.

Many thanks,
Charlie

Matheus-Garbelini, 2022-05-19:
Unfortunately, you won't find (or at least rarely will find) the SoC being used on the product spec documentation.
Instead, you can try searching for the product's FCC listing, which has some tear-down picture of the inside of the product. Unfortunately, now all pictures are clear enough to identify the BT SoC.

Lastly, you can try searching the product on the [Bluetooth Listing Website](https://launchstudio.bluetooth.com/Listings/Search).
Usually, the product lists which chipset is being used in the form of the qualification ID (QID) or combined design ID. You may need to search such IDs associated to the listed product until you can get to a semi. vendor such as Qualcomm, TI, Cypress, etc.

Moreover, you can try buying the same speakers our team was able to test exploit with:
* Mi Portable Bluetooth Speaker - MDZ-36-DB 
* JBL TUNE 500BT  
* Audio Module: XY-WRBT

closed 2022-06-03 by Matheus-Garbelini