Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #8: Building from source

LuckyStripe, 2022-08-07:
Hi there,

I would like to run this on arch Linux as running in a VM is extremely slow and appears to be causing issues.

However I wasn't able to obtain the same versions of dependencies as installed on Ubuntu by the requirements.sh script. When trying to run the `bt_fuzzer` I'm also getting a symbol lookup error in `wdissector/bin/libpagmo.so.6`

So I was looking for a way of recompiling the sources.
I found this in wdissector/docs

```
 1.1) Compile from source (optional)

Several requirements needs to be installed before compiling the project. An automated script for Ubuntu 18.04/20.04 is provided on requirements.sh. To compile from source, simply run the following commands:

git clone https://gitlab.com/asset-sutd/software/wireless-deep-fuzzer -b wdissector --depth=1
cd wdissector
./requirements.sh dev # Install all requirements to compile wdissector from source
./requirements.sh doc  # Install nodejs requirements to generate documentation
./build.sh all # Compile all binaries. It may take around 15min. Go get a coffe!
./build.sh doc # Generate documentation
```
Unfortunately the gitlab project is no longer available and the build script is not part of the `wdissector.tar.zst` file of this repo.

Is there anyway for me to recompile the project on arch or am I forced to set up a machine with Ubuntu?

s-dave, 2022-08-20:
tried it on 18.04. 
It also doesn't work, requirements.sh won't find most of the requirements

Matheus-Garbelini, 2022-10-07:
Hi @LuckyStripe and @s-dave to build from scratch you need access to the source code repo, which is available for Universities students/reasearchers upon request to [https://src.braktooth.com](https://src.braktooth.com).

As for slow VM, meake sure to use VMWare instead of Oracle VM. Another configuration is that you need to select USB 3.0 on the VM usb options.

HangeZyh, 2023-03-27:
Hi @Matheus-Garbelini ,
I also want to request the src, and I have completed the application on this site(https://src.braktooth.com/). 
Sorry to interrupt, but I'd like to know when my application will be approved.
Thx~

matchcase, 2025-02-14:
Hello, I have also completed the application and would like to know when it will be approved.
TIA!

Matheus-Garbelini, 2025-02-14:
@matchcase should be approved now.

flyingduck15, 2025-08-05:
Hey @Matheus-Garbelini I have requested the source code if you could please take a look at my application/request. Thanks!