Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #10: wdissector.tar.zst can't be decompressed

yangting111, 2022-10-07:
[image]

Matheus-Garbelini, 2022-10-07:
Hi @yangting111 have you installed zsyt first?
You can install it with: `sudo apt install zstd` before extracting the file

yangting111, 2022-10-07:
Thank you for your reply！The result is shown as follows：
[image]

Matheus-Garbelini, 2022-10-07:
Thanks @yangting111 
can you try with this instead: https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks/releases/download/v1.0.1/release.zip

After download you can unzip it `unzip release.zip && cd release` then run again `tar -I zstd -xf wdissector.tar.zst`

yangting111, 2022-10-07:
Thank you very much for helping me solve this problem.

closed 2022-10-07 by yangting111