Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #9: Cannot flash the ESP-WROVER-KIT

vincentthong, 2022-10-07:
I am following exactly everything in the readme page.
1) Requirements - done, no problem. Ubuntu 18.04 and ESP-WROVER-KIT-V4.1
2) Installation Instructions

- sudo apt install unzip python3-dev **(ok)**
- unzip esp32driver.zip # Extract esp32driver.zip (firmware package) **(ok)**
- cd release **(ok)**
- python3 firmware.py flash /dev/ttyUSB1 # Please change your serial port to match your ESP32 device. **(failed)**
- I am getting:

running install_scripts
  adding license file "LICENSE" (matched pattern "LICEN[CS]E*")
  error: [Errno 26] Text file busy: 'build/bdist.linux-x86_64/wheel/platformio-6.1.4-py3.6.egg-info'
**and then**
Generating project.checksum
Traceback (most recent call last):
  File "firmware.py", line 181, in <module>
    flash_firmware(args[i + 1])
  File "firmware.py", line 103, in flash_firmware
    generate_project_checksum()
  File "firmware.py", line 53, in generate_project_checksum
    checksum = sha1(get_platformio_version())
  File "firmware.py", line 41, in get_platformio_version
    ['pio', '--version'], stdout=subprocess.PIPE, stderr=subprocess.PIPE).stdout
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pio': 'pio'

I tried holding the boot button and then En button on the KIT, but do not help.
Please advice.

Matheus-Garbelini, 2022-10-07:
Hi @FailFast2022 from the followinf error that you got:
error: [Errno 26] Text file busy

It seems another python program is open?
You can check this by running `ps -ef | grep python` and killing its pid.
Or simply do `sudo pkill -f python`
before running the flash script.

FailFast2022, 2022-10-19:
Thanks Matheus for the advice. 
Sorry for late response.

I found the cause of the issue. 
It was because I am installing the program on a shared drive.
Once I move it to local drive. I am able to install.

closed 2022-10-19 by FailFast2022