Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #12: error unpacking the archive wdissector.tar.zst

mr-slonn, 2022-11-06:
My linux knowledge is 0 :(
When executing point B of the instruction, I get an error, and I can't find a solution

┌──(kali㉿kali)-[~/braktooth_esp32_bluetooth_classic_attacks]
└─$ tar -I zstd -xf wdissector.tar.zst                                                      
zstd: /*stdin*\: unsupported format 
tar: Child returned status 1
tar: Error is not recoverable: exiting now

tar version:
tar (GNU tar) 1.34

zstd version:
*** zstd command line interface 64-bits v1.5.2, by Yann Collet ***

Kali version:
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2022.3
Codename:       kali-rolling

mr-slonn, 2022-11-06:
after much agony, it turned out that the file was not fully downloaded :( (although there were no errors when downloading) and therefore it was not unpacked.

closed 2022-11-06 by mr-slonn

warlock2k, 2022-11-26:
I'm facing the same issue, how did you know that the file wasn't fully downloaded?

mr-slonn, 2022-11-26:
somewhere on the Internet I saw a tip to the question "why does szdt not work?" the answer is "check if the file is damaged when downloading" and what was my surprise that it is 300 megabytes and not 140 kilobytes

warlock2k, 2022-11-27:
@mr-slonn, makes sense. I downloaded the file directly from the releases and it works great now, thank you!