Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #13: bin/bt_fuzzer: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory

mr-slonn, 2022-11-06:
Why can't I execute the command "sudo bin/bt_fuzzer"?

┌──(kali㉿kali)-[~/braktooth_esp32_bluetooth_classic_attacks/wdissector]
└─$ sudo bin/bt_fuzzer --help             
bin/bt_fuzzer: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory

Kali version:
Distributor ID: Kali
Description: Kali GNU/Linux Rolling
Release: 2022.3
Codename: kali-rolling

┌──(kali㉿kali)-[~/braktooth_esp32_bluetooth_classic_attacks/wdissector]
└─$ sudo apt-get install libtbb2 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libtbb2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtbbmalloc2

┌──(kali㉿kali)-[~/braktooth_esp32_bluetooth_classic_attacks/wdissector]
└─$  sudo apt-get install libtbb-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libtbb-dev is already the newest version (2021.7.0-1).
The following packages were automatically installed and are no longer required:
  libwireshark15 libwiretap12 libwsutil13
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1190 not upgraded.

Matheus-Garbelini, 2022-11-08:
Hi, did you run requirements.sh after unpacking the binary package?
Also, I didn't test on kali, so you may get unknown system library errors if not running under Ubuntu 18.04.

mr-slonn, 2022-11-08:
Yes, I do everything according to the instructions. The problem is really in the Kali operating system and the Ubuntu version (it doesn't work in 20.04 either). I installed Ubuntu 18 on a virtual machine and it worked, but there are problems with transferring the device from the host. I will install ubuntu 18 on my laptop and try further :) investigate

closed 2022-11-08 by mr-slonn