Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #14: how to reduce the latency

yangte, 2022-11-10:
Hello ,guys ;

I am having some trouble using these ESP32 development boards, I find the UART latency is 1259us and wonder how to reduce the latency of this serial port?

[ESP32BT] Measuring UART Latency...
[ESP32BT] USB Latency:1259 us [OK]
Serial port /dev/ttyUSB1@4000000 opened

Matheus-Garbelini, 2022-11-10:
Hi, what esp32 board are you using? 
Also are you running ubuntu under a VM or native machine?

yangte, 2022-11-14:
Hi ， @Matheus-Garbelini 
Thanks for your reply .
I use vmware machine and use the ESP-WROVER-E kit , when I use the the vmware I found that my usb latency is 505us . 
I wonder if 505us is enough for these tests ?

Matheus-Garbelini, 2022-11-14:
Hi @yangte.
Yes, 505us is enough. But make sure to allow maximum CPU cores for your vm. For example, a minimum of 4 CPUs is recommended. Using USB 3 (blue port) also helps with latency.

yangte, 2022-11-15:
@Matheus-Garbelini 
Ok , Thanks a lot !

closed 2022-11-15 by yangte