Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #15: ERROR: Could not open /dev/ttyUSB2@115200

mr-slonn, 2022-11-12:
what does it mean in the monitor? What kind of device should be there?

[ESP32BT] HCI Bridge ON: /dev/pts/1
[Monitor] ERROR: Could not open /dev/ttyUSB2@115200

I do according to the instructions from here: https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks/issues/2#issuecomment-1039289299
I run the sudo bin/bt_fuzzer command

Are there video instructions on how and what to use? I'm just a complete 0 in this :(

0xbadd0d0, 2022-12-08:
Hi mr-slonn - did you get it working in the end? [any of the tested attacks work for you?](https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks/issues/18)?

If you have managed to flash the ESP32, you only need the ESP-PROG for the fast serial connection. So you only need 3V3, GND, Rx, Tx pins connected. Don't need the JTAG pins hooked up.

I've noticed that same ERROR line in the logs - but only sometimes.
I think what happens is - one of your earlier attempts failed - and something got stuck - prom on the ESP.
If you unplug-and replug the device, you usually get past that error.

mr-slonn, 2023-05-02:
If I do "`sudo esptool.py erase_flash`" and I see
```
esptool.py v4.5.1
Found 3 serial ports
Serial port /dev/ttyUSB0
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:4b:14:15:c1:40
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 21.1s
Hard resetting via RTS pin...
```

I have to do `sudo python 3 firmware.py flash/dev/ttyUSB0` i.e. use ttyUSB0 ?
Just when I use "`sudo python 3 firmware.py flash /dev/ttyUSB1`" then I get the following message:
```
Serial port /dev/ttyUSB1
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
```
And when I use "`sudo python 3 firmware.py flash /dev/ttyUSB0`" then I get the following message:

```
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
......
Writing at 0x00010000... (7 %)
...
Writing at 0x00040000... (100 %)
Wrote 328416 bytes (199214 compressed) at 0x00010000 in 4.4 seconds (effective 591.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
after_upload(["upload"], [".pio/build/esp32doit-devkit-v1-serial/firmware.bin"])
[Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
==== [SUCCESS] Took 7.15 seconds ====
Environment                 Status    Duration
--------------------------  --------  ------------
esp32doit-devkit-v1-jtag    IGNORED
esp32doit-devkit-v1-serial  SUCCESS   00:00:07.147
=== 1 succeeded in 00:00:07.147 ===
```
I don't understand what this message means:
```
after_upload(["upload"], [".pio/build/esp32doit-devkit-v1-serial/firmware.bin"])
[Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'
```
Successfully or still not successfully I recorded the firmware on ESP32?
Because then the "bin/bt_fuzzer" command does not start, i.e. it starts, but there is an error in the error log as I wrote earlier:

[Monitor] ERROR: Could not open /dev/ttyUSB1@115200
[ESP32BT] Firmware version not detected

Matheus-Garbelini, 2023-05-10:
@mr-slonn 
> [Errno 2] could not open port /dev/ttyUSB1: [Errno 2] No such file or directory: '/dev/ttyUSB1'

You might need to check whether your board really has /dev/ttyUSB1 or just /dev/ttyUSB0
As braktooth is intended to be used with ESP-WROVER-KIT or ESP-ETHERNET-KIT, /dev/ttyUSB1 should always be present.
Linux can also assign a different tty port if you already have something there. Therefore I recommend you check with `dmesg -w` which is the second port added by linux. Your logs mention that you have 3 serial ports, therefore maybe you have something on **/dev/ttyUSB3**?

> Connecting........_____....._____....._____....._____....._____....._____....._____
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

In regards to the `Timed out waiting for packet header` error. In some board and/or computers you have to press and hold the BOOT button of your esp32 board during the flash process.

edatgh, 2023-05-18:
Hi everyone, thank You for Your input in advance!

My environment:
  OS: x86_64 Ubuntu 23.04
  Device: ESP32-WROVER-DEV (ESP32-WROVER-E), Chip: ESP32-D0WDQ6 (Rev 1)

Your comments on below questions/issues/remarks are highly appreciated!:

1. When running the Braktooth BT classic PoC, I'm getting similar errors as already mentioned in the thread, therefore: 
Could someone please elaborate regarding the need/requirement to have exactly ESP-WROVER-KIT or ESP-ETHERNET-KIT hardware and exactly two ttyUSB* devices to get the Braktooth BT classic attacks PoC running?

2. I was able to run the sniffer code OK as per here: https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer; therefore a bit confused, why the Braktooth classic attacks PoC are not working?

3. While reviewing some code and overall description of the Bluetooth Classic Sniffer project, just got the gut-feeling it would be possible to produce the Braktooth BT Classic attacks using the Sniffer project and the ESP32-WROVER-E putting some additional effort, is that correct?