Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #20: xcb io crash

nrathaus, 2023-01-26:
Hi,

I followed the instructions provided but when I run `./bin/bt_fuzzer` I get:

```
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
bt_fuzzer: ../../src/xcb_io.c:673: _XReply: Assertion `!xcb_xlib_extra_reply_data_left' failed.
Aborted
```

Any idea what could be the root cause?

Matheus-Garbelini, 2023-01-27:
Hi @nrathaus what OS are you using?
Is it Ubuntu 18.04?

nrathaus, 2023-01-31:
I was using Ubuntu 20.04 LTS, is there a way you can "future proof" it so that Ubuntu 20.04 or 22.04 (even better) will be supported?

Matheus-Garbelini, 2023-01-31:
@nrathaus try `sudo bin/bt_fuzzer --no-gui`