Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #23: General use issue

0utl4nder, 2023-04-15:
Hello! i tried to follow the guide, and actually i have something. 

Im able to list the exploits

```
> sudo bin/bt_exploiter --list-exploits                                                               
> Enabling Core dump: ulimit -c unlimited
> Loading Model...
> Model Loaded. Total States:169  Total Transistions:1299
> Loop detection ENABLED
> [Modules] Loading C++ Modules...
> [Modules] --> noncompliance_invalid_stop_encryption.so loaded
> [Modules] --> paging_scan_disable.so loaded
> [Modules] --> truncated_sco_link_request.so loaded
> [Modules] --> duplicated_iocap.so loaded
> [Modules] --> sdp_oversized_element_size.so loaded
> [Modules] --> truncated_lmp_accepted.so loaded
> [Modules] --> lmp_max_slot_overflow.so loaded
> [Modules] --> noncomplicance_duplicated_encryption_request.so loaded
> [Modules] --> invalid_timing_accuracy.so loaded
> [Modules] --> wrong_encapsulated_payload.so loaded
> [Modules] --> lmp_auto_rate_overflow.so loaded
> [Modules] --> invalid_feature_page_execution.so loaded
> [Modules] --> au_rand_flooding.so loaded
> [Modules] --> invalid_max_slot.so loaded
> [Modules] --> repeated_host_connection.so loaded
> [Modules] --> lmp_overflow_2dh1.so loaded
> [Modules] --> knob.so loaded
> [Modules] --> invalid_setup_complete.so loaded
> [Modules] --> feature_req_ping_pong.so loaded
> [Modules] --> duplicated_encapsulated_payload.so loaded
> [Modules] --> sdp_unkown_element_type.so loaded
> [Modules] --> feature_response_flooding.so loaded
> [Modules] --> lmp_overflow_dm1.so loaded
> [Modules] --> lmp_invalid_transport.so loaded
> [Modules] 24/24 Modules Compiled / Loaded
> Available Exploits:
> --> 'noncompliance_invalid_stop_encryption'
> --> 'paging_scan_disable'
> --> 'truncated_sco_link_request'
> --> 'duplicated_iocap'
> --> 'sdp_oversized_element_size'
> --> 'truncated_lmp_accepted'
> --> 'lmp_max_slot_overflow'
> --> 'noncomplicance_duplicated_encryption_request'
> --> 'invalid_timing_accuracy'
> --> 'wrong_encapsulated_payload'
> --> 'lmp_auto_rate_overflow'
> --> 'invalid_feature_page_execution'
> --> 'au_rand_flooding'
> --> 'invalid_max_slot'
> --> 'repeated_host_connection'
> --> 'lmp_overflow_2dh1'
> --> 'knob'
> --> 'invalid_setup_complete'
> --> 'feature_req_ping_pong'
> --> 'duplicated_encapsulated_payload'
> --> 'sdp_unkown_element_type'
> --> 'feature_response_flooding'
> --> 'lmp_overflow_dm1'
> --> 'lmp_invalid_transport'
```

but when i try to, for example **scan**  `sudo bin/bt_exploiter --scan` i got this error.
```
Enabling Core dump: ulimit -c unlimited
Loading Model...
Model Loaded. Total States:169  Total Transistions:1299
Loop detection ENABLED
[Modules] Loading C++ Modules...
[Modules] --> noncompliance_invalid_stop_encryption.so loaded
[Modules] --> paging_scan_disable.so loaded
[Modules] --> truncated_sco_link_request.so loaded
[Modules] --> duplicated_iocap.so loaded
[Modules] --> sdp_oversized_element_size.so loaded
[Modules] --> truncated_lmp_accepted.so loaded
[Modules] --> lmp_max_slot_overflow.so loaded
[Modules] --> noncomplicance_duplicated_encryption_request.so loaded
[Modules] --> invalid_timing_accuracy.so loaded
[Modules] --> wrong_encapsulated_payload.so loaded
[Modules] --> lmp_auto_rate_overflow.so loaded
[Modules] --> invalid_feature_page_execution.so loaded
[Modules] --> au_rand_flooding.so loaded
[Modules] --> invalid_max_slot.so loaded
[Modules] --> repeated_host_connection.so loaded
[Modules] --> lmp_overflow_2dh1.so loaded
[Modules] --> knob.so loaded
[Modules] --> invalid_setup_complete.so loaded
[Modules] --> feature_req_ping_pong.so loaded
[Modules] --> duplicated_encapsulated_payload.so loaded
[Modules] --> sdp_unkown_element_type.so loaded
[Modules] --> feature_response_flooding.so loaded
[Modules] --> lmp_overflow_dm1.so loaded
[Modules] --> lmp_invalid_transport.so loaded
[Modules] 24/24 Modules Compiled / Loaded
[ESP32BT] HCI Bridge ON: /dev/pts/5
Setting UART flag ASYNC_LOW_LATENCY
[ESP32BT] Firmware version not detected
Serial port could not open.
BT Scanning Started (Inquiry)...
Setting UART flag ASYNC_LOW_LATENCY
[ESP32BT] Firmware version not detected
Setting UART flag ASYNC_LOW_LATENCY
[ESP32BT] Firmware version not detected
^CMerged logs saved to logs/Bluetooth/logs_merged.txt
Fuzzer Closed
```
i used to had another error which said something like  `Could not open /dev/ttyUSB2@115200`, i used a python module to clear the esp32 and reinstall the firmware `python -m esptool --chip esp32 erase_flash`

im in `6.1.0-kali7-amd64`

Matheus-Garbelini, 2023-04-17:
Hi @0utl4nder , what esp32 baord are you using?
Braktooth only works with ESP-WROVER-KIT or ESP-ETHERNET-KIT

0utl4nder, 2023-04-17:
> Hi @0utl4nder , what esp32 baord are you using?
> Braktooth only works with ESP-WROVER-KIT or ESP-ETHERNET-KIT

Makes sense... Im using a Dev kit v1

closed 2023-05-05 by Matheus-Garbelini