Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #38: code not compiled to run on ARM systems

lachlanngordon, 2024-08-21:
Would you be able to provide a recompiled version of this code that is able to run on ARM architecture?

Matheus-Garbelini, 2024-08-31:
Hi @lachlanngordon. The repo. documentation has been updated with instructions on how to download latest Braktooth release ([v.1.2.0](https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks/releases/tag/v1.2.0)) for ARM64. Note that the release is experimental and might not work well in arm boards due to high process (USB to CPU) latency.

lachlanngordon, 2024-09-12:
Hi @Matheus-Garbelini thank you very much for recompiling for ARM. Would you possibly be able to compile for ARM32 as well please. Thank you so much!

Matheus-Garbelini, 2024-09-17:
Hi @lachlanngordon unfortunately it does not compile for ARM32 due to wireshark compatibility limitations. However, I believe that even if we could compile, Braktooth would run too slow for you to get a good experience with it. Maybe consider using an [Orangepi Zero 2W](https://www.aliexpress.com/item/1005005979335218.html) or [Orange Pi Zero 3](https://www.aliexpress.com/item/1005006500154220.html), which are tiny, cheap and runs aarch64.

Alternatively, you can buy a used Samsung Galaxy 20 FE 5G (~200USD), root and install [nethunter kernel](https://xdaforums.com/t/kernel-nethunter-for-galaxy-s20-fe-5g-r8q-snapdragon.4205881/), which allows you to run docker and access serial ports. In my view if you are working with anything that is not a product and requires portability for research, using a rooted phone is the best approach to run Braktooth in ARM.