Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #44: Segfault when running knob attack

mrmischi3f, 2024-11-25:
I am trying to run the Knob attack using `bt_exploiter`. However when the ESP32 tries to do the connection I get the following:
```
Assigned CPUSET:
0, 1, 2, 3, 
/proc/sys/kernel/sched_rt_runtime_us = -1
Enabling Core dump for this process: ulimit -c unlimited
Loading Model...
Model Loaded. Total States:169  Total Transistions:1299
Loop detection ENABLED
[Modules] Loading C++ Modules at "modules/exploits/bluetooth/"
[Modules] --> repeated_host_connection.so loaded
[Modules] --> duplicated_iocap.so loaded
[Modules] --> paging_scan_disable.so loaded
[Modules] --> truncated_sco_link_request.so loaded
[Modules] --> feature_req_ping_pong.so loaded
[Modules] --> feature_response_flooding.so loaded
[Modules] --> sdp_unkown_element_type.so loaded
[Modules] --> lmp_invalid_transport.so loaded
[Modules] --> truncated_lmp_accepted.so loaded
[Modules] --> lmp_auto_rate_overflow.so loaded
[Modules] --> au_rand_flooding.so loaded
[Modules] --> invalid_max_slot.so loaded
[Modules] --> noncomplicance_duplicated_encryption_request.so loaded
[Modules] --> lmp_max_slot_overflow.so loaded
[Modules] --> lmp_overflow_dm1.so loaded
[Modules] --> invalid_feature_page_execution.so loaded
[Modules] --> knob.so loaded
[Modules] --> sdp_oversized_element_size.so loaded
[Modules] --> duplicated_encapsulated_payload.so loaded
[Modules] --> invalid_setup_complete.so loaded
[Modules] --> wrong_encapsulated_payload.so loaded
[Modules] --> lmp_overflow_2dh1.so loaded
[Modules] --> invalid_timing_accuracy.so loaded
[Modules] --> noncompliance_invalid_stop_encryption.so loaded
[Modules] 24/24 Modules Compiled / Loaded
[ESP32BT] HCI Bridge ON: /dev/pts/3
[ESP32BT] Firmware version: 1.4.1
[ESP32BT] LMP Sniffing ENABLED
[ESP32BT] TX Packet interception ENABLED
[ESP32BT] [!] RX Bypass DISABLED
[ESP32BT] [!] Bypass on Demand DISABLED
[ESP32BT] [!] Role Switch ENABLED
[ESP32BT] Own BDADDR set to bc:bb:b1:8c:dd:4e
[ESP32BT] Measuring UART Latency...
[ESP32BT] USB Latency:119 us [OK]
Serial port /dev/ttyUSB1@4000000 opened
[Monitor] ERROR: Could not open /dev/ttyUSB3@115200
Host BDAddress randomized to e2:d1:b7:c9:c6:f8
[!] Global timeout started with 45 seconds
[BT Program] Starting program bin/sdp_rfcomm_query -u /dev/pts/3 -a b0:be:83:63:92:5d --iocap 3 --authreq 3 --bounding 1
bin/sdp_rfcomm_query -u /dev/pts/3 -a b0:be:83:63:92:5d --iocap 3 --authreq 3 --bounding 1
-u /dev/pts/3 -a b0:be:83:63:92:5d --iocap 3 --authreq 3 --bounding 1
/home/lightship/wdissector/bin/sdp_rfcomm_query
Packet Log: logs/Bluetooth/hci_dump.pklg
H4 device: /dev/pts/3
address=b0:be:83:63:92:5d
iocap=3
authreq=3
bouding=1
Local version information:
- HCI Version    0x0008
- HCI Revision   0x030e
- LMP Version    0x0008
- LMP Subversion 0x030e
- Manufacturer 0x0060
Unknown manufacturer / manufacturer not supported yet.
Local name: 
BTstack up and running on E2:D1:B7:C9:C6:F8.
Starting RFCOMM Query
[Baseband] TX --> FHS
[LMP] TX --> LMP_features_req
[LMP] RX <-- LMP_features_res
[LMP] TX --> LMP_features_req_ext
[LMP] RX <-- LMP_features_res_ext
[LMP] TX --> LMP_features_req_ext
[LMP] RX <-- LMP_features_res_ext
[LMP] TX --> LMP_version_req
[LMP] RX <-- LMP_version_res
[LMP] TX --> LMP_timing_accuracy_req
[LMP] RX <-- LMP_timing_accuracy_res
[LMP] TX --> LMP_host_connection_req
[LMP] RX <-- LMP_slot_offset
[LMP] RX <-- LMP_switch_req
[LMP] TX --> LMP_accepted
[Baseband] RX <-- FHS
[LMP] RX <-- LMP_accepted
[LMP] TX --> LMP_setup_complete
[LMP] RX <-- LMP_setup_complete
[LMP] TX --> LMP_max_slot
[LMP] TX --> LMP_auto_rate
[LMP] RX <-- LMP_packet_type_table_req
[LMP] TX --> LMP_packet_type_table_req
[LMP] RX <-- LMP_max_slot
[LMP] RX <-- LMP_max_slot_req
[LMP] TX --> LMP_accepted
[LMP] RX <-- LMP_auto_rate
[LMP] RX <-- LMP_not_accepted_ext
[LMP] TX --> LMP_accepted_ext
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0041)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0042)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0042)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0043)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0043)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0044)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0044)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0045)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0045)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0046)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0046)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0047)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0047)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0048)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0048)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x0049)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x0049)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x004a)
[L2CAP] RX <-- Connection Response - Refused - no resources available (SCID: 0x004a)
SDP query failed 0x67, retrying...
[L2CAP] TX --> Connection Request (SDP, SCID: 0x004b)
[BT Program] Restart Triggered
Process already started, stopping now...
Segmentation fault
```
From all the tries I've done I conclude that once the SDP query is done the exploiter wants to restart the connection instead of continuing into the pairing phase where the KNOB attack takes place. Am I missing something?

Since the source code of the exploiter is not public, I can not debug exactly what the problem is as the coredump from the segfault indicates some kind of problem related to the threads I assume the binary creates to perform the BT connection.

Can you help me with that?

sacca97, 2025-06-25:
I am not even getting to that point with 1.2.0 release, all I have is:

` ** (wdissector:2191657) [DFilter ERROR] epan/dfilter/dfvm.c:1288 -- dfvm_apply(): assertion failed: tree`

Abin-51, 2025-08-26:
```shell
pu@pu-virtual-machine:~/桌面/wdissector$ sudo gdb bin/bt_exploiter core.30169
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bin/bt_exploiter...
[New LWP 30169]
[New LWP 30205]
[New LWP 30211]
[New LWP 30228]
[New LWP 30208]
[New LWP 30197]
[New LWP 30243]
[New LWP 30198]
[New LWP 30202]
[New LWP 30227]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
--Type <RET> for more, q to quit, c to continue without paging--c
Core was generated by `bin/bt_exploiter --host-port=/dev/ttyUSB1 --target=a4:50:46:8d:f0:0a --exploit='.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007cc10fe92bdd in __pthread_cancel (th=0) at ./nptl/pthread_cancel.c:64
64	./nptl/pthread_cancel.c: 没有那个文件或目录.
[Current thread is 1 (Thread 0x7cc10daa2d00 (LWP 30169))]
(gdb) bt
#0  0x00007cc10fe92bdd in __pthread_cancel (th=0) at ./nptl/pthread_cancel.c:64
#1  0x00007cc11af0ce00 in TinyProcessLib::Process::close_fds (this=0x602e2cd744a0) at /builds/asset-sutd/software/wireless-deep-fuzzer/libs/tinyprocess/process_unix.cpp:298
#2  0x00007cc11af0d10e in TinyProcessLib::Process::kill (this=0x602e2cd744a0, force=true) at /builds/asset-sutd/software/wireless-deep-fuzzer/libs/tinyprocess/process_unix.cpp:359
#3  0x0000602e2266ea52 in ProcessRunner::stop (this=0x602e22788d68 <BTProcess>, force=true, wait=true, autorestart=<optimized out>)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:227
#4  0x0000602e226be092 in ProcessRunner::start_process (this=0x602e22788d68 <BTProcess>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:51
#5  0x0000602e22672ad5 in ProcessRunner::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::function<void (char const*, unsigned long)>, std::function<void (char const*, unsigned long)>, bool) (this=0x602e22788d68 <BTProcess>, 
    process_name="bin/sdp_rfcomm_query", args="-u /dev/pts/3 -a a4:50:46:8d:f0:0a --iocap 3 --authreq 3 --bounding 1", process_callback=..., error_callback=..., 
    change_working_dir=false) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:334
#6  ProcessRunner::init (this=0x602e22788d68 <BTProcess>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:281
#7  0x0000602e226731d2 in ProcessRunner::restart (this=0x602e22788d68 <BTProcess>, force=<optimized out>, wait=<optimized out>, ignore_init_callback=false)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:250
#8  0x0000602e2270ae28 in main::$_4::operator() (reason=<optimized out>, this=<optimized out>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/bt_fuzzer.cpp:1429
#9  main::$_4::__invoke (reason=<optimized out>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/bt_fuzzer.cpp:1425
#10 0x0000602e227064a2 in Machine::RunLoopDetection (wd=0x7cc10da9b1c0, this=<optimized out>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Machine.hpp:860
#11 Machine::RunStateMapper (this=<optimized out>, wd=0x7cc10da9b1c0, new_mapping=false, no_events=false, no_color_change=false, gen_label_only=false)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Machine.hpp:894
#12 0x0000602e226a963c in Machine::RunStateMapper (new_mapping=false, no_events=false, no_color_change=false, this=<optimized out>)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Machine.hpp:958
#13 process_packet (raw_pkt=std::vector of length 15, capacity 28 = {...}, offset=7, dir=1, enable_state_mapper=<optimized out>, duplicated=false, sent_dup_num=0x0)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/bt_fuzzer.cpp:844
#14 0x0000602e22689f38 in main (argc=4, argv=<optimized out>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/bt_fuzzer.cpp:1842
(gdb) bt full
#0  0x00007cc10fe92bdd in __pthread_cancel (th=0) at ./nptl/pthread_cancel.c:64
        pd = <optimized out>
        init_sigcancel = 1
        oldch = <optimized out>
#1  0x00007cc11af0ce00 in TinyProcessLib::Process::close_fds (this=0x602e2cd744a0) at /builds/asset-sutd/software/wireless-deep-fuzzer/libs/tinyprocess/process_unix.cpp:298
No locals.
#2  0x00007cc11af0d10e in TinyProcessLib::Process::kill (this=0x602e2cd744a0, force=true) at /builds/asset-sutd/software/wireless-deep-fuzzer/libs/tinyprocess/process_unix.cpp:359
        lock = {_M_device = @0x602e2cd74528}
#3  0x0000602e2266ea52 in ProcessRunner::stop (this=0x602e22788d68 <BTProcess>, force=true, wait=true, autorestart=<optimized out>)
    at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:227
No locals.
[... cut ...]
#6  ProcessRunner::init (this=0x602e22788d68 <BTProcess>) at /builds/asset-sutd/software/wireless-deep-fuzzer/src/Process.hpp:281
--Type <RET> for more, q to quit, c to continue without paging--

``` 

I also encountered the same issue. When I opened the generated core file with gdb, the reason for the crash of bt_exploiter/bt_fuzzer was basically at the stage of TinyProcessLib::Process::kill() → pthread_cancel(), where the program triggered a SIGSEGV while attempting to stop the child process. However, I'm not quite sure how to make the necessary modifications.