Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #49: [ESP32BT] Firmware Version not Detected

Harsh-bsl, 2025-03-29:
We are using **ESPRESSIF's ESP-WROVER-KIT_V4.1** for Braktooth testing [Detail of the testing-https://github.com/Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks].
Suddenly it stopped working whenever we start Braktooth test. It's seems HW unit not able to detect firmware version. The error message we are seeing are: 
**[ESP32BT] Firmware Version Not Detected**.[Attached image as well]
This issue happened with two HW units. We are unable to do Braktooth testing due to HW unit not working.
 
We tried re-downloading the firmware using command- python3 firmware.py flash /dev/ttyUSB1. But the issue not resolved.

[image]

Harsh-bsl, 2025-03-29:
Hi @Matheus-Garbelini ,
Caould you please help us here ?

Harsh-bsl, 2025-04-04:
Hi @Matheus-Garbelini ,

Need your kind help? Could you please me with the above error?

Matheus-Garbelini, 2025-04-05:
Hi @Harsh-bsl try adding the command line to force which serial port to use when running braktooth binary:
`sudo bin/bt_fuzzer --host-port=/dev/ttyUSB1`
You host-port should be the same port you used for flashing the board. It might change if you connect other peripherals with serial port on your laptop, so feel free to try different ports. It's not clear from your logs which port is being used. You can try attaching here the logs in the braktooth `logs` folder.

Harsh-bsl, 2025-04-07:
Hi @Matheus-Garbelini ,

Thanks for the inputs. I confirmed that the right USB port should be USB1 from the dmesg logs.
As soon as i plug in micro-USB and plug in ESP-WROVER-KIT_V4.1 , below prints appear in the dmesg logs-

[image]

Below is the output if i give the command _**sudo bin/bt_fuzzer --host-port=/dev/ttyUSB1**_ -

[image]

It worked previously with USB1 properly, somehow not working suddenly, didn't know what happen-

[image]

[image]

Attaching logs as-

[logs_bk.zip](https://github.com/user-attachments/files/19634852/logs_bk.zip)

Please do let me know if you need any more info in details.

Thanks,
Harsh

sut4nd0, 2025-04-24:
Hi, May I ask if you have solved this problem?