Matheus-Garbelini/braktooth_esp32_bluetooth_classic_attacks #53: Making the code work on aarch64 Ubuntu 24.04

XenoKovah, 2025-05-25:
I'd like to use Braktooth on an Ubuntu 24.04 ARM (AARCH64) VM on a Mac, and so far I had to do the following:

Find and replace in ./requirements.sh:
`libtbb2` -> `libtbbmalloc2`
`libasound2` -> `libasound2t64`
Then add `libgflags-dev` and `libgoogle-glog-dev` to the list.

However I'm still encountering the following issue:

```
$ sudo ./bin/bt_exploiter 
./bin/bt_exploiter: error while loading shared libraries: libglog.so.0: cannot open shared object file: No such file or directory
```

It seems that `libgoogle-glog-dev ` doesn't provide `libglog.so.0`? I haven't been able to find the correct replacement for this yet. Any ideas? (Do you need to compile a binary specifically for U24.04 to deal with its changed libraries? Or can this requirement be removed?)

shipcod3, 2025-07-18:
I got the same issue with the first question.

mrheinen, 2025-08-07:
The glog issue can be overcome with:

sudo cp /usr/lib/x86_64-linux-gnu/libglog.so /usr/lib/x86_64-linux-gnu/libglog.so.0

But there are other libraries, at least on Debian, which are outdated and replaced. Even if you install the newer versions and try the hacky approach of copying the libs to the expected names; you will run into issues.

I requested source code access and hope I get it in order to contribute.

mrheinen, 2025-08-09:
Awesome. I'll have a look starting next week!

orionparker33, 2026-10-18:
Hi @Matheus-Garbelini — I'd like to request source access as well, for the same reason as @mrheinen: to get the aarch64 build working and contribute the fix back.

About me: I'm an offensive security researcher (CISSP) running a private Bluetooth/RF test bench. Not academic — I saw the request form is for university emails, so I'm asking here instead. Use is internal lab work and contributing fixes; no redistribution.

Setup: Raspberry Pi 5 (aarch64, kernel 6.18), two ESP32-Ethernet-Kit-VE boards flashed from `release/` with esptool (the PlatformIO `toolchain-xtensa32` has no linux_aarch64 package, as noted above), `wdissector_aarch64` 1.2.0 with glog/ssl1.1 shims so all 24 modules compile and load.

Concrete bug I can work on: `bt_exploiter` 1.2.0 arm64 segfaults reproducibly the moment the serial link to the board comes up and the helper (`bin/sdp_rfcomm_query`) is started. gdb on the core: `ProcessRunner::start_process` → start callback (`bt_fuzzer.cpp:1719`) → `PacketLogger::writeLog` (`PacketLogger.hpp:155`) → `memcpy` with a corrupted length (the upper 32 bits are lost — looks like a size truncation that's benign on x86_64). Happy to share the core/backtrace and submit a patch once I can build.

If you can send a GitLab invite, my email is **orionparker33@gmail.com**. Thanks for keeping this project alive.